arduino/toolchain-avr #17: Update Avrdude to 6.1

karlbackstrom, 2015-01-07:
Would it be possible to ask for an update to Avrdude 6.1 (from 6.0.1)? All the current patches seems to apply quite cleanly and there are some features in 6.1 that would be great to have for external platforms (especially the flip2 programmer for Xmega chips).

PeterVH, 2015-06-03:
+1 for the upgrade. It will help ArduinoISP: 6.1 implements some clever fixes that make it possible to program high addresses (over 128KB) with the stk500v1 protocol (which is what ArduinoISP uses). This makes it possible to reliably burn the bootloader of an atmega2560 using ArduinoISP.

gh-megabit, 2015-07-10:
:+1: against this, too. In particular, AVRdude 6.1 adds support for Atmel-ICE, the current-generation Atmel ISP device.

matthijskooijman, 2015-12-09:
6.2 has been released by now, too.

Testato, 2015-12-09:
+1

ArduinoISP fir bootloading the Mega is a very old problem, finally avrdude 6.1 sokve the problem.
And the AtmelICE support is also important

After avrdude update will be possible official insert AtmelICE programmer in the IDE

ffissore, 2015-12-09:
I'm working on avrdude 6.2

facchinm, 2015-12-09:
avrdude 6.2 is affected by [this](https://lists.gnu.org/archive/html/avrdude-dev/2015-12/msg00004.html) bug, dropping the update until it gets fixed

NicoHood, 2015-12-09:
Could you also apply this patch then:
https://github.com/arduino/Arduino/issues/3645

I still dont have an avrdude account. I tried to contact atmel 3 times about 16u2 datasheet faults, but they do not respond (or the wrong person answers). They just dont want to fix the datasheet. But that should not keep us from fixing avrdude, so could anyone post this information to joerg wunsch?

gh-megabit, 2016-01-03:
I've found avrdude 6.1 and 6.2 to be unreliable with my Atmel-ice - and that's after dealing with the LibUSB filter mess. It might be better to look at another tool for this programmer - atprogram is proving reliable and is easy to enable for uploads. The burn bootloader option may need code changes. I'm not sure if this utility is available on OS X or Linux.

facchinm, 2016-01-04:
AFAIK `atprogram` only runs on Windows, not a big deal. `avrdude 6.1` works great with Atmel-ICE (tested) and will be integrated in the IDE as soon as we update it officially

PaoloP74, 2016-01-08:
Can you publish a version 1.6.7-R1?

PaoloP74, 2016-03-15:
AVRdude **6.3** is out.
--> http://mirror.lihnidos.org/GNU/savannah//avrdude/avrdude-doc-6.3.pdf

cmaglie, 2016-07-27:
Avrdude 6.3 build scripts are now on their own repository:

https://github.com/arduino/avrdude-build-script

closed 2016-07-27 by cmaglie