arduino/toolchain-avr #18: Add to avrdude.conf

PaoloP74, 2015-02-16:
Can you add these definitions to the file avrdude.conf?

On "ATtiny85" section

>#------------------------------------------------------------
>#ATtiny85
>#------------------------------------------------------------
>
>part
>    pagel            = 0xB3;
>     bs2              = 0xB4;
>
>memory "lock"
>     read            = "0 1 0 1  1 0 0 0  0 0 0 0  0 0 0 0",
>                           "0 0 0 0  0 0 0 0  o o o o  o o o o";
>

--> http://forum.arduino.cc/index.php?topic=197790.msg1462378#msg1462378

cmaglie, 2015-09-09:
Usually we don't fork or add patches to tools like avrdude unless it's absolutely necessary.
This looks like a bug that must be notified to upstream avrdude's developers so they can validate it and integrate in the next release (and consequently we can push it to arduino too).

What does these definitions means? What kind of bug they solve?
Can you check if this change is already present in the avrdude repository but not yet released?
http://svn.savannah.nongnu.org/viewvc/trunk/avrdude/avrdude.conf.in?root=avrdude&view=markup

PaoloP74, 2015-09-09:
This definition
>#------------------------------------------------------------
>#ATtiny85
>#------------------------------------------------------------
>
>memory "lock"
>     read            = "0 1 0 1  1 0 0 0  0 0 0 0  0 0 0 0",
>                           "0 0 0 0  0 0 0 0  o o o o  o o o o";
>

is used to set the fuse "memory lock" in ATtiny85. Without the definition avrdude goes wrong.
>Reading |                                                    | 0% 0.00savr_read(): error reading address 0x0000
>    read operation not supported for memory "lock"
>avrdude: failed to read all of lock memory, rc=-2

The secon definition
>part
>pagel = 0xB3;
>bs2 = 0xB4;

is to prevent the warning when uploading code.
If avrdude, in the latest versions, does not find these configuration reports a warnings but continues. In previous versions the violation stops the upload.