arduino/toolchain-avr #22: Regression: avrdude Linux build fails to use USBtinyISP programmer with libusb error

tdicola, 2015-04-04:
This is a regression in the latest Arduino build for Linux (1.6.3) and the avrdude binary it uses that was built from this toolchain.  If you use the USBtinyISP programmer the call to avrdude fails with this error:
```
avrdude: error: no usb support. Please compile again with libusb installed.
```

An easy way to repro this without having to use any special hardware is to download an Arduino IDE build and run this command from its hardware/tools/avr/bin directory:
```
./avrdude -c usbtiny -p attiny85 -C ../etc/avrdude.conf
```

In this latest 1.6.3 release you'll see the command immediately fail with this error:
```
tony@tony-main:~/Downloads/arduino-1.6.3/hardware/tools/avr/bin$ ./avrdude -c usbtiny -p attiny85 -C ../etc/avrdude.conf
avrdude: error: no usb support. Please compile again with libusb installed.
avrdude: programmer operation not supported

avrdude done.  Thank you.
```

However the same command in the 1.6.1 release works fine and fails with the expected error that it can't find the attached USBtinyISP programmer (i.e. the code detects it is successfully built/linked against libusb):
```
tony@tony-main:~/Programs/arduino-1.6.1/hardware/tools/avr/bin$ ./avrdude -c usbtiny -p attiny85 -C ../etc/avrdude.conf
avrdude: Error: Could not find USBtiny device (0x1781/0xc9f)

avrdude done.  Thank you.
```

I have a strong suspicion that the recent change to use static linking on linux (this commit: https://github.com/arduino/toolchain-avr/commit/c503eaa8f8578b650cc2b064fd1d0a15bf6d24f6) caused this problem.  I'm not familiar enough with avrdude's build process, but perhaps it's missing a flag or parameter to tell it to statically link in libusb?

For the impact of this issue it is actually somewhat severe since the USBtinyISP programmer now won't work with the avrdude shipped by the Arduino IDE on Linux.  Some custom boards like Adafruit's Trinket and Gemma (and soon Arduino's Gemma) use a bootloader that mimics the USBtinyISP and are also affected by this issue and broken on Linux.

Note that if you want to repro this using the Arduino IDE itself, be sure to first delete your ~/arduino15/ cache folder.  I found there's some other unrelated bug in the IDE that appears to be using a stale or older avrdude build from that cache folder.  I.e. if you have 1.6.1 installed then upgrade to 1.6.3 you might be fine because the IDE is still using the old 1.6.1 avrdude.  However if you delete the cached folder or do a fresh install of the IDE on a machine then you get the latest avrdude with this libusb error.

ffissore, 2015-04-14:
Tag https://github.com/arduino/toolchain-avr/releases/tag/avr-gcc-4.8.1-avrdude-6.0.1-arduino4 should fix it. Sent too early: an updated toolchain will be soon released

closed 2015-04-14 by ffissore

ffissore, 2015-04-23:
1.6.4 core and tools (with libusb fix) available through boards manager