arduino/toolchain-avr #23: Avrdude on Linux64 is compiled whitout libusb

ffissore, 2015-04-10:
_From @eni23 on April 9, 2015 20:5_

arduino: 1.6.3 release from arduino-website
os: debian wheezy amd64

the avrdude-binary is not compiled with libusb, which means no usb-serial adapters are supported.   since most arduino-boards depend on usb-communication this is a quite big impact. 
workaround: compile avrdude 6.0.1 on your own and replace broken binary's

_Copied from original issue: arduino/Arduino#2952_

ffissore, 2015-04-10:
_From @mschlenker on April 10, 2015 6:57_

USB serial adapters usually are supported by the linux kernels drivers, not userland drivers. Those userland drivers are usually just needed for more exotic programmers. 

Could you please post the USB IDs of the USB serial adapter (lsusb output) and the kernel version you are using (uname -a). Then after plugging in the usb serial adapter take a look at the last lines in the out put of dmesg.

ffissore, 2015-04-10:
_From @matthijskooijman on April 10, 2015 7:15_

Agreed that libusb isn't needed for usb-serial adapaters, but not having libusb is still a problem for non-serial programmers like the JTAGICE3 and some others.

ffissore, 2015-04-10:
I've checked the logs, and indeed since we switched to static compiling, avrdude is not linking to libusb anymore. Problem is without static, gcc fails with a missing LTO plugin.

eni23, 2015-04-10:
u i was a little bit in a hurry i meant isp programmers and not uart-adapters. but anyway, it should be possible to compile it static with libusb..

tdicola, 2015-04-12:
If it helps, here's a similar issue I raised about this where USBtinyISP programming (for boards like Trinket, Gemma, or the real USBtinyISP programmer) doesn't work because of the static link missing libusb issue: https://github.com/arduino/toolchain-avr/issues/22

larsemil, 2015-04-14:
Yes. programming Gemma is not possible with 1.6.3.

`avrdude was compiled without usb support.`

ffissore, 2015-04-14:
Working on this atm

ffissore, 2015-04-14:
Tag https://github.com/arduino/toolchain-avr/releases/tag/avr-gcc-4.8.1-avrdude-6.0.1-arduino4 should fix it

ffissore, 2015-04-14:
Sent too early: an updated toolchain will be soon released

closed 2015-04-14 by ffissore

matthijskooijman, 2015-04-21:
What Arduino IDE will this upodated toolchain be released in? 1.6.4 I presume?

ffissore, 2015-04-22:
Thanks to boards manager, we are going to release a 1.6.4 avr core very soon. May be even tomorrow CET morning

ffissore, 2015-04-23:
1.6.4 core and tools (with libusb fix) available through boards manager

shimanyi, 2015-08-27:
@ffissore Any quick fixes on this? I am embarrassingly new in this, I get the error mentioned /bin/avrdude: not found but I can't get any fixes on this. Please give me pointers. I just need to upload from sublime.

Thanks

ffissore, 2015-08-28:
@shimanyi your issue has nothing to do with this one. This is about missing support to libusb. Your issue is about a non existent /bin/avrdude

shimanyi, 2015-08-28:
@ffissore Thanks for the feedback, how do I get around that though?

ffissore, 2015-08-28:
I don't know. It's an issue with stino

shimanyi, 2015-08-28:
Ok, thanks. Will keep looking.