arduino/toolchain-avr #25: Provide fully static binaries

facchinm, 2015-07-21:
arduino/Arduino#3548 and other issue are frequently caused by the binaries not being fully static. 

Consider providing static binaries next time the programmers will be updated

facchinm, 2015-09-09:
Small update:
- OSX programs can't be fully static (http://stackoverflow.com/questions/5259249/creating-static-mac-os-x-c-build)
- Linux build need to recompile glibc entirely ( :dizzy_face: ) but it is ok
- Windows build does not need to be static as it loads the dll from the executable folder

ffissore, 2015-09-09:
I missed this issue. We tried and we failed. `-static` on linux was introduced with c503eaa8f8578b650cc2b064fd1d0a15bf6d24f6 and reverted in cc809be997faf8f31c7c433acd69fb0a214baebd because of #23

facchinm, 2015-09-09:
Linux script here: https://gist.github.com/facchinm/6c3feff432c62d6849fb

bam80, 2015-11-21:
How it going? ARDUINO 1.6.6 is out and again - no static binaries..

zoobab, 2016-01-05:
I am having the same problem running the Arduino software on Gentoo:

"/home/zoobab/Downloads/arduino-1.6.7/hardware/tools/avr/bin/avrdude: error while loading shared libraries: libtinfo.so.5: cannot open shared object file: No such file or directory"

I will try to compile the arduino SDK with MUSL-Libc as GLIBC is a nightmare to produce static binaries. GLIBC guys deliberately broke that feature for GPL/political reasons.

facchinm, 2016-01-05:
Hi @zoobab ,
the problem you are experiencing is due to a newly compiled `libncurses`. Symlinking `libtinfo` to `libncurses` or installing some kind of `libtinfo-compat ` package does the trick.
However, the next version of `avrdude` binary will be of course static to avoid this kind of problems.

zoobab, 2016-01-05:
I know, I had to rebuild ncurses with tinfo support, so now at least I can upload some code to my arduino.

Will try to build a static toolchain now, will write here if I manage to build it.

facchinm, 2016-01-05:
No need for a static toolchain, only a static avrdude is needed.
If you want to compile using this scripts https://gist.github.com/facchinm/6c3feff432c62d6849fb and validate them it would be great!

ShorTie8, 2016-01-26:
From my Googling, static is dead.
that is why ' -static ' doesn't work, or more like ignored any more.

zoobab, 2016-01-26:
Not really, the "-static" is being broken on purpose by the GLIBC guys. I will check what I can do, even if it means using MUSL libc like I did for other projects where I needed a static binary.

NicoHood, 2016-07-04:
The question is (for me as packager), if we just can use the upstream avrdude etc or if this would cause too many problems. The same goes for java. However I still suggest to use static linkage as the ready to go zipfile should not rely on any dependencies.

facchinm, 2016-07-19:
will be solved by https://github.com/arduino/avrdude-build-script/commits/master

NicoHood, 2016-07-19:
@facchinm Since it is very close to the upstream avrdude, do you think in arch we just can use the upstream avrdude? Also what do you think of using a different avr-gcc? It might break a few programs possibly, but also it can make use of lto etc. I also found some libusb stuff, can we also get rid of this in the arduino build process if the host has those available?

facchinm, 2016-07-19:
Using non-bundled tools (in a transparent way) is a support nightmare; people should also know the tools version which is not directly obtainable by the UI. 
But the linked avrdude (and all the toolchain from https://github.com/arduino/Arduino/pull/5021) would solve these problem since:
* binaries are almost static (libusb stuff/additional libraries are linked statically)
* lto is enabled
* the toolchains are Atmel's official

So I'd go for the bundled ones :smile:

NicoHood, 2016-07-19:
Thanks for explaining this. That's also what I was thinking.

Is the patch in the latest master build now? I am creating an arduino-git package, so I could remove the avrdude dep.

cmaglie, 2016-07-27:
https://github.com/arduino/Arduino/pull/5021 is now merged and released

closed 2016-07-27 by cmaglie