arduino/toolchain-avr #27: Upgrading, More official versions ??

ShorTie8, 2015-12-29:
Been playing, And was wondering if it would help to use some of the more "Official" versions of avr-binutils, avr-gcc and avr-gdb along with adding atleast ncurses ??

Like from:  http://distribute.atmel.no/tools/opensource/Atmel-AVR-GNU-Toolchain/3.5.0/

Or is there something that all the patching is doing special ??

cmaglie, 2015-12-30:
It depends on what you mean with "official", because there is the "official" upstream avr-gcc http://www.nongnu.org/avr-libc/ and various different patched release around. The most valuable "patched" version is the one made by Atmel, in fact our current release tracks the Atmel-AVR-GNU-Toolchain 3.4.5, the patching that you see are the diffs between the upstream avr-gcc and the Atmel patched gcc.

In other words what the build script does is:

- Download gcc tarballs (and other tools as well) from upstream development site
- Apply patches (to make it equal to Atmel release)
- Build

Since Atmel is out with 3.5.0 we can now consider upgrading the toolchains to this release.

cmaglie, 2016-07-27:
We are now tracking directly Atmel bundle.

closed 2016-07-27 by cmaglie