arduino/toolchain-avr #30: mpc link offline

NicoHood, 2016-01-18:
https://github.com/arduino/toolchain-avr/blob/master/gcc.build.bash#L46

+ [[ ! -f mpc-0.9.tar.gz ]]
+ wget http://www.multiprecision.org/mpc/download/mpc-0.9.tar.gz
--2016-01-17 23:24:18--  http://www.multiprecision.org/mpc/download/mpc-0.9.tar.gz
Resolving www.multiprecision.org (www.multiprecision.org)... 46.30.212.205
Connecting to www.multiprecision.org (www.multiprecision.org)|46.30.212.205|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-01-17 23:24:18 ERROR 404: Not Found.

NicoHood, 2016-01-19:
I used the waybackmachine's download and the toolchain compiled fine again.

cmaglie, 2016-02-03:
IIRC this was a sf.net mirror, maybe we should host a mirror of the files on the arduino.cc server and keep the original link in the comments.

NicoHood, 2016-02-03:
sf.net?

Hosting a copy would make sense. But it would also be nice to get an overview of all available arduino downloads from the arduino server. Could you make this also available somehow?

cmaglie, 2016-07-27:
> sf.net?

sourceforge.net

BTW I was wrong is not a mirror but the official repository of mpc, that quickly went back online.

At this point I prefer to not copy these files but use the official repo.

closed 2016-07-27 by cmaglie