arduino/toolchain-avr #36: Support for ATmega328PB

awatterott, 2016-07-26:
The latest Atmel toolchain does not support the new ATmega328**PB**, but the needed files are available from the [ATmega Series Device Support (1.0.98)](http://packs.download.atmel.com).
Can you please add the files (header file, device-specs, avrdude device description) to the next toolchain release?
Here is also an updated header file with the old register/bit naming:
https://github.com/watterott/ATmega328PB-Testing/tree/master/hardware/tools/avr

mvcl, 2016-10-04:
Please include support for the Atmega328PB support going into the releases. This is a very useful chip, it works well, and has some really helpful added capabilities. I especially appreciate having the second Serial channel. 

Watterott's contributed changes make the ATmega328PB work very well.

gohai, 2017-05-08:
Please include support for Atmega328PB.

nielsnl68, 2017-07-26:
What is the status of this request? I like to start using this chip very soon.

facchinm, 2017-07-26:
Next toolchain will include 328pb support for sure, alongside 3.6.0 toolchain

closed 2017-08-02 by awatterott