arduino/toolchain-avr #38: Some additional information on building the toolchain

morrissinger, 2016-07-30:
Would be great for newbs to understand more about what is required to build the toolchain. The Readme lists the four components of the toolchain, but says nothing of what to do with the various scripts after cloning the repo.

For example, we could spell out that one needs to:

  1. ./tools.bash
  2. /binutils.build.bash
  3. ./gcc.build.bash
  4. ./avr-libc.build.bash

Etc., and to specify an order to the build.

morrissinger, 2016-07-31:
On examining the scripts, it looks like running `./package-avr-gcc.bash` would probably yield the entire toolchain?

One thing we could do is just note this in the Readme. I'd make the PR myself, but I am not yet sufficiently confident in this repo's codebase to make PRs. Would love it if someone who knows this repo could ensure that this information is accurate.

NicoHood, 2016-09-18:
This is really missing in the readme. It took me some time to find this stuff.

NicoHood, 2016-09-18:
I cannot compile the toolchain, it exits with this error:
```
g++ -c  -DIN_GCC_FRONTEND -w -O2 -g0  -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -Icp -I../../gcc/gcc -I../../gcc/gcc/cp -I../../gcc/gcc/../include -I../../gcc/gcc/../libcpp/include -I/home/arch/toolchain-avr/gcc-build/./gmp -I/home/arch/toolchain-avr/gcc/gmp -I/home/arch/toolchain-avr/gcc-build/./mpfr -I/home/arch/toolchain-avr/gcc/mpfr -I/home/arch/toolchain-avr/gcc/mpc/src  -I../../gcc/gcc/../libdecnumber -I../../gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc/gcc/../libbacktrace    -o cp/except.o -MT cp/except.o -MMD -MP -MF cp/.deps/except.TPo ../../gcc/gcc/cp/except.c
In file included from ../../gcc/gcc/cp/except.c:1013:0:
cfns.gperf: In function ‘const char* libc_name_p(const char*, unsigned int)’:
cfns.gperf:101:1: error: ‘const char* libc_name_p(const char*, unsigned int)’ redeclared inline with ‘gnu_inline’ attribute
cfns.gperf:26:14: note: ‘const char* libc_name_p(const char*, unsigned int)’ previously declared here
make[2]: *** [Makefile:1057: cp/except.o] Error 1
make[2]: *** Waiting for unfinished jobs....
rm gcov.pod cpp.pod gfdl.pod gcc.pod fsf-funding.pod
make[2]: Leaving directory '/home/arch/toolchain-avr/gcc-build/gcc'
make[1]: *** [Makefile:3959: all-gcc] Error 2
make[1]: Leaving directory '/home/arch/toolchain-avr/gcc-build'
make: *** [Makefile:846: all] Error 2
```

Any ideas why?

morrissinger, 2016-09-18:
@NicoHood: what is your environment? What OS?

facchinm, 2016-09-19:
Hi @NicoHood , I believe you are on Arch with gcc 6.1 or newer. The bug has been reported [here](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=69959) and the fix is [here](https://gcc.gnu.org/viewcvs/gcc?view=revision&revision=233720).
We are not applying it because we are compiling against older targets (read: Debian7) for maximum compatibility

KarlNorway, 2017-01-23:
Im having difficulties getting the texinfo 4+ installed with Cygwin, the only option is 6.+. Also all bash files seems to have wrong line endings. i get this when I try to run (any) bash:
`
./build.conf:  line 3: $\r : could not find command 
`

Also is it one option to "just " copy the precompiled dll in the AVR tool chain?

ReachableCEO, 2021-05-27:
Reversed (or previously applied) patch detected!  Assume -R? [n]
Apply anyway? [n]
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file gcc/configure.ac.rej
patching file gcc/doc/invoke.texi
Reversed (or previously applied) patch detected!  Assume -R? [n]
Apply anyway? [n]
Skipping patch.
4 out of 4 hunks ignored -- saving rejects to file gcc/doc/invoke.texi.rej
patching file gcc/ira.c
Reversed (or previously applied) patch detected!  Assume -R? [n]
Apply anyway? [n]
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file gcc/ira.c.rej
patching file gcc/testsuite/gcc.c-torture/compile/limits-externdecl.c
Reversed (or previously applied) patch detected!  Assume -R? [n]
Apply anyway? [n]
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file gcc/testsuite/gcc.c-torture/compile/limits-externdecl.c.rej
patching file gcc/testsuite/lib/target-supports.exp
Hunk #1 succeeded at 7966 (offset 18 lines).
patching file gcc/tree.h
Reversed (or previously applied) patch detected!  Assume -R? [n]
Apply anyway? [n]
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file gcc/tree.h.rej
+ mv gmp-5.0.2 gcc/gmp
mv: cannot stat 'gmp-5.0.2': No such file or directory

This is on Ubuntu 20.04 x86-64.