arduino/toolchain-avr #39: Reliance on MacPorts?

morrissinger, 2016-07-30:
We could rely on Homebrew rather than on MacPorts for wget, automake and autoconf. These packages are available via Homebrew, which does not require installing an additional package manager. Additionally, the reliance on MacPorts can present problems for people on certain widespread ISPs, running vendor-provided equipment, where rsync may be blocked.

morrissinger, 2016-08-01:
One possible solution is to build the toolchain inside a VM. I did that and made it available here: https://atlas.hashicorp.com/morrissinger/boxes/arduino/

nathansizemore, 2016-09-27:
> These packages are available via Homebrew, which does not require installing an additional package manager.

Homebrew is not a macOS application, so using it also requires an additional install from a base macOS system.

morrissinger, 2016-09-27:
@nathansizemore: noted. However, the latter point is still an issue.

cmaglie, 2016-09-28:
The use of macports is just a suggestion not a mandatory dependency.

The real dependency is on the tools, `wget`, `automake`, etc. no matter how you install them.

maybe we should make this more explicit in the readme?