arduino/toolchain-avr #40: Build toolchain from source

nathansizemore, 2016-09-27:
It would be nice to be able to rely on individual toolchains built from source, with a few shell scripts, rather than relying on a release from Atmel, then applying patches. 

Taking advantage of "new", post 2011 standards as default, along with added benefits of various additional sanitizer flags, and alias optimizations. 

Is there any thought in making this a possibility, along with maintenance, or is this a pipe dream that will never happen?

cmaglie, 2016-09-28:
> It would be nice to be able to rely on individual toolchains built from source, with a few shell scripts, rather than relying on a release from Atmel, then applying patches.

Uhm I don't understand, actually we **are** relying on toolchains built from source using [**a few shell scripts**](https://github.com/arduino/toolchain-avr/blob/master/package-avr-gcc.bash) to build them.

The patch we make to Atmel's toolchain are really small (just one or two lines) and ATM their purpose is only to make it compile on our environments. 

> Taking advantage of "new", post 2011 standards as default, along with added benefits of various additional sanitizer flags, and alias optimizations.

avr-gcc 4.9.2 has been release on Q4-2014, maybe not bleeding edge technology, but stable. We prefer less features but stable and reliable.

nathansizemore, 2016-09-28:
> Uhm I don't understand, actually we are relying on toolchains built from source using a few shell scripts to build them.

Should have been clearer, I meant providing toolchain building from released source, instead of from Atmel. Following GCC and friend's source mirrors. This would remove a dependency. Currently it seems it is GNU release -> Atmet release -> Arduino bundled.

closed 2016-09-28 by nathansizemore