arduino/toolchain-avr #42: Missing Support for Tiny 4/5/9/10/20 (Reduced Core TinyAVR)

sleemanj, 2016-12-01:
The avr-gcc 4.9.2 in IDE 1.6.13 does still not know about the ATTiny 4/5/9/10/20 Reduced Core TinyAVR chips.

The [official avr-gcc 4.9.2](http://www.atmel.com/tools/ATMELAVRTOOLCHAINFORLINUX.aspx) does

Official
---
    $ avr-gcc -Wa,-mlist-devices --target-help 2>&1 | grep "tiny4 "
    atxmega128a1u atxmega128a4u attiny4 attiny5 attiny9 attiny10 attiny20

Arduino 1.6.13
---
    $ avr-gcc -Wa,-mlist-devices --target-help 2>&1 | grep "tiny4 "
    ( Nothing Found )

sleemanj, 2016-12-01:
I think the problem is that the build process isn't bringing in the "tinyavr" gcc core for some reason.  I can't see  a configuration option readily, maybe this is a discrepancy with the GPL released sources not matching the Atmel (Microchip) build of gcc?

Official Build
---
    avr8-gnu-toolchain-linux_x86_64$ ./bin/avr-gcc --target-help | grep avrtiny
                   avrtiny   - AVR Tiny core with 16 gp registers
    avrxmega2 avrxmega3 avrxmega4 avrxmega5 avrxmega6 avrxmega7 avrtiny
    avrtiny: 

IDE 1.6.13 Build
---
    arduino-1.6.13/hardware/tools/avr$ ./bin/avr-gcc --target-help  | grep avrtiny
    (Nothing found)

sleemanj, 2016-12-02:
It seems that building avr-gdb nukes the avrtiny chips.  

Doing a grep for "avtriny" in the `objdir`

Before Building gdb
---
    Binary file avr/bin/ld matches
    Binary file avr/bin/ld.bfd matches
    Binary file avr/bin/as matches
    Binary file bin/avr-c++ matches
    Binary file bin/avr-as matches
    Binary file bin/avr-ld matches
    Binary file bin/avr-ld.bfd matches
    Binary file bin/avr-g++ matches
    Binary file bin/avr-gcc matches
    Binary file bin/avr-cpp matches
    Binary file bin/avr-gcc-4.9.2 matches
    lib/gcc/avr/4.9.2/device-specs/specs-attiny4:# Auto-generated specs for AVR device attiny4 (core avrtiny, 16-bit SP)
    lib/gcc/avr/4.9.2/device-specs/specs-attiny4: -mmcu=avrtiny
    lib/gcc/avr/4.9.2/device-specs/specs-attiny4: %{!mmcu=avr*: %<mmcu=* -mmcu=avrtiny} %<msp8
    lib/gcc/avr/4.9.2/device-specs/specs-avrtiny:# Auto-generated specs for AVR core architecture avrtiny
    lib/gcc/avr/4.9.2/device-specs/specs-avrtiny: -mmcu=avrtiny
    lib/gcc/avr/4.9.2/device-specs/specs-attiny40:# Auto-generated specs for AVR device attiny40 (core avrtiny, 16-bit SP)
    lib/gcc/avr/4.9.2/device-specs/specs-attiny40:  -mmcu=avrtiny
    lib/gcc/avr/4.9.2/device-specs/specs-attiny40:  %{!mmcu=avr*: %<mmcu=* -mmcu=avrtiny} %<msp8
    lib/gcc/avr/4.9.2/device-specs/specs-attiny10:# Auto-generated specs for AVR device attiny10 (core avrtiny, 16-bit SP)
    lib/gcc/avr/4.9.2/device-specs/specs-attiny10:  -mmcu=avrtiny
    lib/gcc/avr/4.9.2/device-specs/specs-attiny10:  %{!mmcu=avr*: %<mmcu=* -mmcu=avrtiny} %<msp8
    lib/gcc/avr/4.9.2/device-specs/specs-attiny5:# Auto-generated specs for AVR device attiny5 (core avrtiny, 16-bit SP)
    lib/gcc/avr/4.9.2/device-specs/specs-attiny5: -mmcu=avrtiny
    lib/gcc/avr/4.9.2/device-specs/specs-attiny5: %{!mmcu=avr*: %<mmcu=* -mmcu=avrtiny} %<msp8
    lib/gcc/avr/4.9.2/device-specs/specs-attiny9:# Auto-generated specs for AVR device attiny9 (core avrtiny, 16-bit SP)
    lib/gcc/avr/4.9.2/device-specs/specs-attiny9: -mmcu=avrtiny
    lib/gcc/avr/4.9.2/device-specs/specs-attiny9: %{!mmcu=avr*: %<mmcu=* -mmcu=avrtiny} %<msp8
    lib/gcc/avr/4.9.2/device-specs/specs-attiny20:# Auto-generated specs for AVR device attiny20 (core avrtiny, 16-bit SP)
    lib/gcc/avr/4.9.2/device-specs/specs-attiny20:  -mmcu=avrtiny
    lib/gcc/avr/4.9.2/device-specs/specs-attiny20:  %{!mmcu=avr*: %<mmcu=* -mmcu=avrtiny} %<msp8
    Binary file lib/gcc/avr/4.9.2/avrtiny/libgcc.a matches
    Binary file lib/gcc/avr/4.9.2/avrtiny/libgcov.a matches
    Binary file libexec/gcc/avr/4.9.2/lto1 matches
    Binary file libexec/gcc/avr/4.9.2/cc1 matches
    Binary file libexec/gcc/avr/4.9.2/cc1plus matches
    share/info/gcc.info:     'avrtiny'
    share/info/gcc.info:     for MCU='avrtiny', 'avrxmega2', 'avrxmega4', 'avrxmega5',
    Binary file share/info/bfd.info matches
    share/info/as.info:     Instruction set avrtiny is for the ATtiny4/5/9/10/20/40
    share/man/man1/avr-g++.1:.ie n .IP """avrtiny""" 4
    share/man/man1/avr-g++.1:.el .IP "\f(CWavrtiny\fR" 4
    share/man/man1/avr-g++.1:.IX Item "avrtiny"
    share/man/man1/avr-g++.1:for \fImcu\fR=\f(CW\*(C`avrtiny\*(C'\fR, \f(CW\*(C`avrxmega2\*(C'\fR, \f(CW\*(C`avrxmega4\*(C'\fR,
    share/man/man1/avr-gcc.1:.ie n .IP """avrtiny""" 4
    share/man/man1/avr-gcc.1:.el .IP "\f(CWavrtiny\fR" 4
    share/man/man1/avr-gcc.1:.IX Item "avrtiny"
    share/man/man1/avr-gcc.1:for \fImcu\fR=\f(CW\*(C`avrtiny\*(C'\fR, \f(CW\*(C`avrxmega2\*(C'\fR, \f(CW\*(C`avrxmega4\*(C'\fR,
    x86_64-pc-linux-gnu/avr/include/bfd.h:#define bfd_mach_avrtiny   100

After Building gdb
---
    Binary file bin/avr-c++ matches
    Binary file bin/avr-g++ matches
    Binary file bin/avr-gcc matches
    Binary file bin/avr-cpp matches
    Binary file bin/avr-gcc-4.9.2 matches
    lib/gcc/avr/4.9.2/device-specs/specs-attiny4:# Auto-generated specs for AVR device attiny4 (core avrtiny, 16-bit SP)
    lib/gcc/avr/4.9.2/device-specs/specs-attiny4: -mmcu=avrtiny
    lib/gcc/avr/4.9.2/device-specs/specs-attiny4: %{!mmcu=avr*: %<mmcu=* -mmcu=avrtiny} %<msp8
    lib/gcc/avr/4.9.2/device-specs/specs-avrtiny:# Auto-generated specs for AVR core architecture avrtiny
    lib/gcc/avr/4.9.2/device-specs/specs-avrtiny: -mmcu=avrtiny
    lib/gcc/avr/4.9.2/device-specs/specs-attiny40:# Auto-generated specs for AVR device attiny40 (core avrtiny, 16-bit SP)
    lib/gcc/avr/4.9.2/device-specs/specs-attiny40:  -mmcu=avrtiny
[... 4 lines cut ...]
    lib/gcc/avr/4.9.2/device-specs/specs-attiny5:# Auto-generated specs for AVR device attiny5 (core avrtiny, 16-bit SP)
    lib/gcc/avr/4.9.2/device-specs/specs-attiny5: -mmcu=avrtiny
    lib/gcc/avr/4.9.2/device-specs/specs-attiny5: %{!mmcu=avr*: %<mmcu=* -mmcu=avrtiny} %<msp8
    lib/gcc/avr/4.9.2/device-specs/specs-attiny9:# Auto-generated specs for AVR device attiny9 (core avrtiny, 16-bit SP)
    lib/gcc/avr/4.9.2/device-specs/specs-attiny9: -mmcu=avrtiny
    lib/gcc/avr/4.9.2/device-specs/specs-attiny9: %{!mmcu=avr*: %<mmcu=* -mmcu=avrtiny} %<msp8
    lib/gcc/avr/4.9.2/device-specs/specs-attiny20:# Auto-generated specs for AVR device attiny20 (core avrtiny, 16-bit SP)
    lib/gcc/avr/4.9.2/device-specs/specs-attiny20:  -mmcu=avrtiny
    lib/gcc/avr/4.9.2/device-specs/specs-attiny20:  %{!mmcu=avr*: %<mmcu=* -mmcu=avrtiny} %<msp8
    Binary file lib/gcc/avr/4.9.2/avrtiny/libgcc.a matches
    Binary file lib/gcc/avr/4.9.2/avrtiny/libgcov.a matches
    Binary file libexec/gcc/avr/4.9.2/lto1 matches
[... cut ...]
    share/man/man1/avr-g++.1:.IX Item "avrtiny"
    share/man/man1/avr-g++.1:for \fImcu\fR=\f(CW\*(C`avrtiny\*(C'\fR, \f(CW\*(C`avrxmega2\*(C'\fR, \f(CW\*(C`avrxmega4\*(C'\fR,
    share/man/man1/avr-gcc.1:.ie n .IP """avrtiny""" 4
    share/man/man1/avr-gcc.1:.el .IP "\f(CWavrtiny\fR" 4
    share/man/man1/avr-gcc.1:.IX Item "avrtiny"
    share/man/man1/avr-gcc.1:for \fImcu\fR=\f(CW\*(C`avrtiny\*(C'\fR, \f(CW\*(C`avrxmega2\*(C'\fR, \f(CW\*(C`avrxmega4\*(C'\fR,
    x86_64-pc-linux-gnu/avr/include/bfd.h:#define bfd_mach_avrtiny   100

sleemanj, 2016-12-02:
Eureka!  `make install` in gdb.build.bash needs to be `make install-gdb`  

For the reason we can look in the [official build script](http://distribute.atmel.no/tools/opensource/Atmel-AVR-GNU-Toolchain/3.5.4/build-avr8-gnu-toolchain-git.sh) 

    task_start "Installing GDB for $1... "
     # New versions of gdb share the same configure/make scripts with binutils. Running make install-gdb to
     # install just the gdb binaries.
     make $PARALLEL_JOBS install-gdb > make.install.out 2>&1 || task_error "Installing GDB for $1 failed. See make.install.out." "make.install.out"
   
In short, make install for (avr-)gdb will also install another, older, version of binutils, which doesn't have the proper support for the avrtiny (4/5/9/10 etc Reduced Core TinyAVR) in it.

cmaglie, 2016-12-02:
fixed by #43

closed 2016-12-02 by cmaglie