arduino/toolchain-avr #48: [OSX] toolchain 3.6.0 produces wrong binary with 328p target

facchinm, 2018-02-13:
Moved from https://github.com/arduino/Arduino/issues/6899

The issue looks OSX specific; to reproduce, install the staging avr core (1.6.200) and compile AnalogReadSerial. The result will contain a `?` at the end of every println.

facchinm, 2018-02-16:
Should be fixed by AVR core 1.6.205 (based on `staging` branch)

MCUdude, 2018-02-18:
I assume you want the OSX toolchain discussion to continue here.  
Here's my comment from #47:  
  
  
> I just installed 1.6.205 and compiled AnalogReadSerial for ATmega328B using [MiniCore](https://github.com/MCUdude/MiniCore/tree/PB_support). Same error when using [Watterott's 328PB core](https://github.com/watterott/ATmega328PB-Testing).

> It compiles fine on ATmega328P and ATmega48PB/88PB/168PB. Seems like the hardware isn't defined when compiling for 328PB. Here's the error:

>Output before the error occurs:
```
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/Hans/Library/Arduino15/packages -hardware /Users/Hans/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/Hans/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/Hans/Documents/Arduino/libraries -fqbn=MiniCore:avr:328:bootloader=true,variant=modelPB,BOD=2v7,LTO=Os,clock=16MHz_external -ide-version=10805 -build-path /var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_678556 -warnings=all -build-cache /var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_cache_435973 -prefs=build.warn_data_percentage=75 -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/AnalogReadSerial/AnalogReadSerial.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/Hans/Library/Arduino15/packages -hardware /Users/Hans/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/Hans/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/Hans/Documents/Arduino/libraries -fqbn=MiniCore:avr:328:bootloader=true,variant=modelPB,BOD=2v7,LTO=Os,clock=16MHz_external -ide-version=10805 -build-path /var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_678556 -warnings=all -build-cache /var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_cache_435973 -prefs=build.warn_data_percentage=75 -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/AnalogReadSerial/AnalogReadSerial.ino
Using board '328' from platform in folder: /Users/Hans/Documents/Arduino/hardware/MiniCore/avr
Using core 'MCUdude_corefiles' from platform in folder: /Users/Hans/Documents/Arduino/hardware/MiniCore/avr
Build options changed, rebuilding all
Detecting libraries used...
"/Users/Hans/Library/Arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino1/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_ATmega328 -DARDUINO_ARCH_AVR   "-I/Users/Hans/Documents/Arduino/hardware/MiniCore/avr/cores/MCUdude_corefiles" "-I/Users/Hans/Documents/Arduino/hardware/MiniCore/avr/variants/pb-variant" "/var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_678556/sketch/AnalogReadSerial.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/Users/Hans/Library/Arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino1/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_ATmega328 -DARDUINO_ARCH_AVR   "-I/Users/Hans/Documents/Arduino/hardware/MiniCore/avr/cores/MCUdude_corefiles" "-I/Users/Hans/Documents/Arduino/hardware/MiniCore/avr/variants/pb-variant" "/var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_678556/sketch/AnalogReadSerial.ino.cpp" -o "/var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_678556/preproc/ctags_target_for_gcc_minus_e.cpp"
"/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_678556/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/Users/Hans/Library/Arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino1/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_ATmega328 -DARDUINO_ARCH_AVR   "-I/Users/Hans/Documents/Arduino/hardware/MiniCore/avr/cores/MCUdude_corefiles" "-I/Users/Hans/Documents/Arduino/hardware/MiniCore/avr/variants/pb-variant" "/var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_678556/sketch/AnalogReadSerial.ino.cpp" -o "/var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_678556/sketch/AnalogReadSerial.ino.cpp.o"
```

>Error:
 ```
In file included from /Users/Hans/Library/Arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino1/avr/include/avr/pgmspace.h:90:0,
                 from /Users/Hans/Documents/Arduino/hardware/MiniCore/avr/cores/MCUdude_corefiles/Arduino.h:28,
                 from /var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_678556/sketch/AnalogReadSerial.ino.cpp:1:
[... cut ...]
/Applications/Arduino.app/Contents/Java/examples/01.Basics/AnalogReadSerial/AnalogReadSerial.ino: In function 'void setup()':
AnalogReadSerial:16: error: 'Serial' was not declared in this scope
   Serial.begin(9600);
   ^
/Applications/Arduino.app/Contents/Java/examples/01.Basics/AnalogReadSerial/AnalogReadSerial.ino: In function 'void loop()':
AnalogReadSerial:24: error: 'Serial' was not declared in this scope
   Serial.println(sensorValue);
   ^
exit status 1
'Serial' was not declared in this scope
```

facchinm, 2018-02-19:
Hi @MCUdude , indeed the OSX build had a problem importing the right `.h` in `io.h` due to a bash version mismatch. I'm fixing it and publishing a new toolchain with the correct content.

MCUdude, 2018-02-19:
Great! Will test it tonight 👍

MCUdude, 2018-02-19:
I get this error when trying to install 1.6.206:

[image]

per1234, 2018-02-19:
It looks like the checksum value in the JSON file wasn't updated. It should be `7e5440b0902733f82956c89e554106f25c2ad540ac36286ccba3ceac785bcbba`, not `bb179e353c96e9f2ca904d785e4e15aede16a15b3957c7e087f62bd6960d4b83`.

It's strange that when updating from 1.6.206 you get that less helpful message:
```
 Error downloading http://downloads.arduino.cc/cores/avr-1.6.21.tar.bz2
java.lang.RuntimeException: java.lang.Exception: Error downloading http://downloads.arduino.cc/cores/avr-1.6.21.tar.bz2
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:176)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading http://downloads.arduino.cc/cores/avr-1.6.21.tar.bz2
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:131)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:69)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:102)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:173)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 416
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:194)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:129)
	... 4 more
```
But on a fresh install you get the more useful message:
```
 CRC doesn't match, file is corrupted. It may be a temporary problem, please retry later.
java.lang.RuntimeException: java.lang.Exception: CRC doesn't match, file is corrupted. It may be a temporary problem, please retry later.
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:176)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: CRC doesn't match, file is corrupted. It may be a temporary problem, please retry later.
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:91)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:102)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:173)
	... 1 more
```

facchinm, 2018-02-19:
Fixed, thank you for spotting this.

MCUdude, 2018-02-19:
AnalogReadSerial now compiles and runs fine on ATmega328PB. Will do some more testing to verify that everything works fine with the 328PB 🙂

closed 2018-04-17 by facchinm