arduino/toolchain-avr #58: Switch OSX build to x86_64

SConaway, 2019-01-12:
As you've probably heard, Apple is deprecating x32 software soon. What's needed to switch to x86_64 for this toolchain? I've tried building it on my mac, but to no avail.

facchinm, 2019-01-14:
avr-gcc 7.3.0 (compiled using staging branch) (https://github.com/arduino/Arduino/issues/7949#issuecomment-442969335) already targets x86_64 on OSX (https://github.com/arduino/toolchain-avr/commit/df66a5f11d4a5b32b267c3ad88ef519f4d6221c1).
Can you test that build (or try to reproduce it) and report if the warning is not triggered anymore? Thanks!

SConaway, 2019-01-15:
Trying. It's probably not going to show a warning if x32, but I can run` file *.`

SConaway, 2019-02-11:
@facchinm  when is this getting released?

closed 2019-06-23 by SConaway