arduino/toolchain-avr #60: Update toolchain to include ATmega_DFP 1.2.209

MCUdude, 2019-02-20:
As I've mentioned in other threads I'm working on a new Arduino core that (hopefully) will support the new MegaAVR-0 series. I got proper test hardware for all targets (ATmega3808/ATmega4808 and ATmega3209/4809). However during testing I discovered that iom3208.h and iom3209.h lacks a few register definitions and support macros like USART and timers. 

 In various documents Microchip/Atmel states that the ATmega3209 is directly compatible with ATmega4809. The _only_ difference is the flash and ram size. To quote the intro of chapter two in the [megaAVR-0 manual](http://ww1.microchip.com/downloads/en/DeviceDoc/40002015A.pdf):

> Vertical migration [between 3209/4809 and 3208/4808] is possible without code modification, as these devices are fully pin and feature compatible.

At the moment this is not possible by using the Arduino toolchain. It turns out that the iom3208.h and iom3209.h files are missing register definitions and support macros. [The datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/40002016A.pdf) states the 3209 has four UARTs, but only two are actually available. Timer3 (TCB3) is not available either, even though it should. On 3208 USART2 is not defined either.

These errors were later fixed by Microchip, and the latest "ATmega patch" (ATmega_DFP 1.2.209) seems to be correct.

It would be great if the Arduino toolchain was updated with ATmega_DFP 1.2.209, as this would make cross target support much more smooth. I'd _love_ to have an experimental Arduino toolchain available with ATmega_DFP 1.2.209 patched in (and maybe the ATtiny_DFP 1.3.229 too? 😉)

facchinm, 2019-02-21:
@MCUdude I pushed https://github.com/arduino/toolchain-avr/commit/b235983972887ac66fee36b0753e8a17285726e6 and the builds are running :slightly_smiling_face: . I'll post the link as soon as they are ready so you can validate them.
About attiny, it doesn't look difficult; if I prepare some more builds could you validate them too?

MCUdude, 2019-02-21:
Great! I'd test it this afternoon.

I'm not able to compile anything Arduino related for the new ATtinys yet (main priority is the megaAVR-0 series), but I'd definitly have a look at it anyways!

facchinm, 2019-02-21:
Links are
http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-armhf-pc-linux-gnu.tar.bz2
http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-aarch64-pc-linux-gnu.tar.bz2
http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-x86_64-apple-darwin14.tar.bz2
http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-i686-w64-mingw32.zip
http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-i686-pc-linux-gnu.tar.bz2
http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-x86_64-pc-linux-gnu.tar.bz2

Let me know if they fix the 3809 issues!

MCUdude, 2019-02-21:
Thanks! However, how do I make Arduino IDE point to this version? I'm on macOS BTW. Last time I tested beta toolchains you(?) provided a JSON boards file.

facchinm, 2019-02-21:
You can grab the tarball for mac, extract it and replace all the files that currently live in `/Users/yourusername/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel...` with the content of the archive (keeping the same directly name for the base folder, so `...arduino4`, otherwise the builder could complain :wink: )

MCUdude, 2019-02-21:
Something can't be right here. I get this error in the IDE:
`fork/exec /bin/avr-g++: no such file or directory`

Seems like avr-g++ and some other files are just symbolic links.

[image]

facchinm, 2019-03-04:
Mmmh, it could be a problem with normal extractors (compared with IDE one). I'll setup an additional json soon (before the end of today probably).

facchinm, 2019-03-04:
Here we go  http://downloads.arduino.cc/packages/package_avr_7.3.0_staging_index.json

MCUdude, 2019-03-04:
Thanks! You'll hear from me soon

MCUdude, 2019-03-04:
Nope, I still have lots of aliases..

[image]

[image]

facchinm, 2019-03-06:
mmmh ok, I don't think it's a problem if they are linking to the right executable... Did you try compiling something with the new core?

MCUdude, 2019-03-06:
I'm not at my home computer at the moment, but I did try to compile. I got an error saying that avr-g++ wasn't found. I also tried double-clicking on the aliases, but I also got an error saying that the file they're pointing to doesn't exist.

freemovers, 2019-03-07:
I am very excited to see development in making these new MCU's compatible with the Arduino. I can't wait to start projects with the TinyAVR-0. I noticed that on this last build the ATtiny AVR-0 are in the device-specs, but they are not listed in the io.h (yet?), and I don't know which IO file in the include folder should be used for the ATtiny (if any) or where these could be found.  Does Atmel (Microchip) have these in their device packs for Atmel Studio (or that is probably issue MCUdude had).

freemovers, 2019-03-07:
Found the iotn1616.h in the device pack, and added a reference to this file in the IO.h Seems to work now, just have to update the timers in the variant files now. Will start testing soon.

facchinm, 2019-03-08:
I **should** have fixed the toolchain to also contain new Attiny devices.
The json is always http://downloads.arduino.cc/packages/package_avr_7.3.0_staging_index.json and also the core version is the same, so the only way to apply the update if you already have core 1.6.220 is to uninstall and reinstall it.

MCUdude, 2019-03-08:
Well, I'm sorry to disappoint you, but I see no improvements..
Installed 1.6.220 and compiled a blank sketch with Arduino Uno WiFi Rev2 as target (ArduinoCore-megaavr)

```
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/Hans/Library/Arduino15/packages -hardware /Users/Hans/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/Hans/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/Hans/Documents/Arduino/libraries -fqbn=arduino:megaavr:uno2018:mode=on -ide-version=10808 -build-path /var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_84318 -warnings=all -build-cache /var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_cache_888051 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/Users/Hans/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino14 -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=/Users/Hans/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino14 -prefs=runtime.tools.arduinoOTA.path=/Users/Hans/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Users/Hans/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avr-gcc.path=/Users/Hans/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/Users/Hans/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -verbose /var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/untitled1730278803.tmp/sketch_mar08a/sketch_mar08a.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/Hans/Library/Arduino15/packages -hardware /Users/Hans/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/Hans/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/Hans/Documents/Arduino/libraries -fqbn=arduino:megaavr:uno2018:mode=on -ide-version=10808 -build-path /var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_84318 -warnings=all -build-cache /var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_cache_888051 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/Users/Hans/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino14 -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=/Users/Hans/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino14 -prefs=runtime.tools.arduinoOTA.path=/Users/Hans/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Users/Hans/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avr-gcc.path=/Users/Hans/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/Users/Hans/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5 -verbose /var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/untitled1730278803.tmp/sketch_mar08a/sketch_mar08a.ino
Using board 'uno2018' from platform in folder: /Users/Hans/Library/Arduino15/packages/arduino/hardware/megaavr/1.6.220
Using core 'arduino' from platform in folder: /Users/Hans/Library/Arduino15/packages/arduino/hardware/megaavr/1.6.220
Build options changed, rebuilding all
Detecting libraries used...
/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DUNO_WIFI_REV2_328MODE -I/Users/Hans/Library/Arduino15/packages/arduino/hardware/megaavr/1.6.220/cores/arduino/api/deprecated -I/Users/Hans/Library/Arduino15/packages/arduino/hardware/megaavr/1.6.220/cores/arduino -I/Users/Hans/Library/Arduino15/packages/arduino/hardware/megaavr/1.6.220/variants/uno2018 /var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_84318/sketch/sketch_mar08a.ino.cpp -o /dev/null
fork/exec /bin/avr-g++: no such file or directory
Error compiling for board Arduino Uno WiFi Rev2.

```

facchinm, 2019-03-08:
It simply may be a problem with platform.txt, maybe the compiler {runtime.path} is trying to use the old toolchain. I'd take a look but I'm on the go, checking on Monday :)

freemovers, 2019-03-11:
I was able to compile an empty sketch for the ATtiny1616 after making some changes to the core (mainly in the timers), but I am now stuck on the linker. I get the following error:
`"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-gcc-ar" rcs "C:\\Users\\USERNAME\\AppData\\Local\\Temp\\arduino_build_571184\\core\\core.a" "C:\\Users\\USERNAME\\AppData\\Local\\Temp\\arduino_build_571184\\core\\wiring_shift.c.o"
Archiving built core (caching) in: C:\Users\USERNAME\AppData\Local\Temp\arduino_cache_561418\core\core_arduino_megaavr_at1616_4bf7e7c892683359d7a10a2f29691bf6.a
Linking everything together...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-gcc" -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -Wl,--section-start=.text=0x200 -mmcu=attiny1616 -o "C:\\Users\\SVANDE~1\\AppData\\Local\\Temp\\arduino_build_571184/ATtiny1616.ino.elf" "C:\\Users\\USERNAME\\AppData\\Local\\Temp\\arduino_build_571184\\sketch\\ATtiny1616.ino.cpp.o" "C:\\Users\\USERNAME\\AppData\\Local\\Temp\\arduino_build_571184\\core\\variant.c.o" "C:\\Users\\USERNAME\\AppData\\Local\\Temp\\arduino_build_571184/core\\core.a" "-LC:\\Users\\USERNAME\\AppData\\Local\\Temp\\arduino_build_571184" -lm
c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: cannot find crtattiny1616.o: No such file or directory

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: cannot find -lattiny1616

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board ATtiny1616.`

Is there a specific linker configuration file in the variants folder?

facchinm, 2019-03-11:
@freemovers it looks like you are still using the old compiler (5.4.0) since all the paths point to the normal Arduino installation. To make sure the new one gets used, modify `plaftorm.txt` to read `compiler.path={runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path}/bin/`

freemovers, 2019-03-11:
Seems the compiler and linker are working now, but they linker is now throwing and error that the libgcc, libm and libc are incompatible (testing it with the "Arduino Uno WIFI Rev2"):
```
C:\Users\username\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5\bin>avr-gcc -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -Wl,--section-start=.text=0x200 -mmcu=atmega4809 -o "C:\\Users\\SVANDE~1\\AppData\\Local\\Temp\\arduino_build_529723/sketch_mar11a.ino.elf" "C:\\Users\\SVANDE~1\\AppData\\Local\\Temp\\arduino_build_529723\\sketch\\sketch_mar11a.ino.cpp.o" "C:\\Users\\SVANDE~1\\AppData\\Local\\Temp\\arduino_build_529723\\core\\variant.c.o" "C:\\Users\\SVANDE~1\\AppData\\Local\\Temp\\arduino_build_529723/core\\core.a" "-LC:\\Users\\SVANDE~1\\AppData\\Local\\Temp\\arduino_build_529723" -lm
c:/users/username/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: skipping incompatible c:/users/username/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib\libm.a when searching for -lm
c:/users/username/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: cannot find -lm
c:/users/username/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: skipping incompatible c:/users/username/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/../lib/gcc/avr/7.3.0\libgcc.a when searching for -lgcc
c:/users/username/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: cannot find -lgcc
c:/users/username/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: skipping incompatible c:/users/username/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib\libm.a when searching for -lm
c:/users/username/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: cannot find -lm
c:/users/username/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: skipping incompatible c:/users/username/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib\libc.a when searching for -lc
c:/users/username/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: cannot find -lc
collect2.exe: error: ld returned 1 exit status
```

Any other changes required to the platform or board file? Looks like the actual files are there.

facchinm, 2019-03-12:
On Linux everything compiles correctly with the new toolchain and UNO WIFI rev2 as target, I'm checking on WIndows

facchinm, 2019-03-12:
@freemovers confirmed, the Windows toolchain had a bug; now it should be fixed. Please uninstall core 1.6.220 and reinstall it in a couple of hours (the time for the updated toolchain to distribute across all the CDN)

freemovers, 2019-03-12:
I can confirm that it all works now. I was able to compile the Arduino Uno Wifi Rev2 (ATmega4809) and the ATtiny1616 with AVR-GCC 7.3.0 from Arduino. I will start working on the variance files for the ATtiny series.