arduino/toolchain-avr #63: Arduino IDE - internal compiler error: Segmentation fault

paulr4, 2019-06-19:
I'm suffering from the problem raised in [this forum post](https://forum.arduino.cc/index.php?topic=572746.0), resulting in a segmentation fault in the compiler. It goes away when downgrading to Arduino boards package 6.1.21 from 6.1.23.

Posts there indicate it's been fixed already, but is a release to the IDE anywhere on the horizon?
Sorry if this is the wrong place to ask, I'm not sure what else to do!

facchinm, 2019-06-19:
This is the proposed PR that includes all the latest goodies to fix toolchain 5.4.0 bugs  https://github.com/arduino/Arduino/pull/8976 .
The issue has been reported a lot of times so I'm closing as duplicate :wink:

closed 2019-06-19 by facchinm

paulr4, 2019-06-19:
Thanks for the response, is it going to see the light of day soon or does it need further testing?

facchinm, 2019-06-20:
It still needs testing before becoming official, but reporting on that thread if it fixes your problems and eventually other info will speedup the merge.

paulr4, 2019-06-24:
I would be very happy to test it, but I'm afraid I don't know how... is there a guide anywhere that you can point me toward?

per1234, 2019-06-24:
@paulr4 

1. Open this link: https://github.com/arduino/Arduino/pull/8976#issuecomment-503638485
2. You will see a reply from ArduinoBot with a list of links to downloads of the test build of the Arduino IDE for each operating system. Click on the link for the operating system you're using.
3. Wait for the download to finish.
4. Extract the downloaded file
5. Create a folder inside the unzipped folder named `portable`. This will cause the test build of the Arduino IDE to run in [portable mode](https://www.arduino.cc/en/Guide/PortableIDE) so it uses the new version of Arduino AVR Boards instead of the 1.6.21 version you installed.
6. You will find the arduino executable inside the extracted folder. Run that to start the Arduino IDE.

paulr4, 2019-06-24:
Thanks, I'll give it a go!

paulr4, 2019-06-24:
I'm afraid the segmentation fault bug is very fickle and I couldn't recreate it even on the old build. If I run into it again I'll test the new build then.