arduino/toolchain-avr #65: ATTiny10 and Toolchain 7.3.0

wholder, 2019-11-15:
Arduino recently upgraded the AVR toolchain in the macOS version from 32 bit 5.4.0 to 64 bit 7.3.0 in order to support macOs Catalina, which is 64 bit only.  However, the  7.3.0 toolchain has an odd problem when trying to link code compiled for the attiny10 series of microcontrollers.  To investigate this, I downloaded the Mac version of the toolchain from:

  http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-x86_64-apple-darwin14.tar.bz2
 
And used it to compile this simple test program:

```
#include <avr/io.h>
int main () {
    DDRB |= (1 << PB0);
  while (true) {
        PORTB |= (1 << PB0);
        PORTB &= ~(1 << PB0);
    }
}
``` 
I can use the 7.3.0 toolchain to compile and link this code for the attiny85, like this:
 
```
  ./avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -mmcu=**attiny85** -I ./ test.cpp -o test.cpp.o
  ./avr/bin/avr-gcc -w -Os -g -mmcu=**attiny85** -o test.elf test.cpp.o -L ./ -lm 

``` 
and it works just fine.  But, if i change the -**mmcu** switch from **attiny85** to **attiny10**, the compile seems to work fine, but the command to link the code (2nd line) spews out the following errors:
 
```
/Users/wholder/Desktop/temp/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: cannot find crtattiny10.o: No such file or directory
/Users/wholder/Desktop/temp/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: skipping incompatible /Users/wholder/Desktop/temp/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/libm.a when searching for -lm
/Users/wholder/Desktop/temp/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: cannot find -lm
/Users/wholder/Desktop/temp/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: skipping incompatible /Users/wholder/Desktop/temp/avr/bin/../lib/gcc/avr/7.3.0/libgcc.a when searching for -lgcc
/Users/wholder/Desktop/temp/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: cannot find -lgcc
/Users/wholder/Desktop/temp/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: skipping incompatible /Users/wholder/Desktop/temp/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/libm.a when searching for -lm
/Users/wholder/Desktop/temp/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: cannot find -lm
/Users/wholder/Desktop/temp/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: skipping incompatible /Users/wholder/Desktop/temp/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/libc.a when searching for -lc
/Users/wholder/Desktop/temp/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: cannot find -lc
/Users/wholder/Desktop/temp/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: cannot find -lattiny10
collect2: error: ld returned 1 exit status

``` 
Curiously, the file crtattiny10.o is available in the toolchain at path: ./avr/lib/avrtiny/crtattiny10.o
 
Can anyone help me figure out what's going on here?
 
Wayne

facchinm, 2019-11-29:
**Following up from the [mailing list](https://groups.google.com/a/arduino.cc/forum/?utm_medium=email&utm_source=footer#!msg/developers/Ny82s7t11Vc/tqOAPj35AwAJ), better to keep these things near the code**

I investigated a bit and the problem is:
* atpacks don't provide `libgcc`, which gets generated by the toolchain build phase, but only `libc` and `libm`
* specifying `avrtiny` folder with `-L` finds at least compatible `libc` and `crt*`

Said that, there isn't much we can do except waiting for the devices to be upstreamed. As @WestfW correctly noted,  Microchip only provides GCC 5.4.0 as "latest", but we know it has tons of issues with lto so we couldn't really use their infrastructure...

enekochan, 2020-07-22:
I was having the same problem using the toolchain (version 7.3.0-atmel3.6.1-arduino7) provided by Arduino 1.9.0 (beta). Then I downloaded the [toolchain from Microchip](https://www.microchip.com/mplab/avr-support/avr-and-arm-toolchains-c-compilers) and it worked fine. Hope this helps.

wholder, 2020-07-30:
Thanks or the pointer to the Microchip toolchains!

Wayne

WestfW, 2024-09-15:
I have been looking into this recently, not that I've come up with anything substantial.
I'll note that the Microchip distribution of avr-gcc IS now updated to 7.3, and supports the avrtiny devices by default (without needing to resort to definitions in a "pack")

I am unable to find any suspicious differences between the Microchip gcc 7.3 and the Arduino gcc 7.3 :-(
The libraries are there, but for some reason link isn't finding them.  Looking at "verbose" output from the compile, compared with a successful build, I see the "avrtiny" directories are missing from the library paths, and there's no **-imultilib avrtiny** in the cc1plus command.   But I don't know why...

Microchip avr-g++
```
Using built-in specs.
Reading specs from /usr/local/avr8-Microchip-3.7.0/bin/../lib/gcc/avr/7.3.0/device-specs/specs-attiny10
COLLECT_GCC=avr-g++
COLLECT_LTO_WRAPPER=/usr/local/avr8-Microchip-3.7.0/bin/../libexec/gcc/avr/7.3.0/lto-wrapper
Target: avr
Configured with: /Volumes/build_disk/jenkins_mcu/workspace/avr8-gnu-toolchain-osx/src/gcc/configure LDFLAGS=-L/Volumes/build_disk/jenkins_mcu/workspace/avr8-gnu-toolchain-osx/avr8-gnu-toolchain-darwin_x86_64-hostlibs/lib CPPFLAGS= --target=avr --host=x86_64-apple-darwin17.0.0 --build=x86_64-apple-darwin17.0.0 --prefix=/Volumes/build_disk/jenkins_mcu/workspace/avr8-gnu-toolchain-osx/avr8-gnu-toolchain-darwin_x86_64 --libdir=/Volumes/build_disk/jenkins_mcu/workspace/avr8-gnu-toolchain-osx/avr8-gnu-toolchain-darwin_x86_64/lib --enable-languages=c,c++ --with-dwarf2 --enable-doc --disable-shared --disable-libada --disable-libssp --disable-nls --with-avrlibc=yes --with-mpfr=/Volumes/build_disk/jenkins_mcu/workspace/avr8-gnu-toolchain-osx/avr8-gnu-toolchain-darwin_x86_64-hostlibs --with-gmp=/Volumes/build_disk/jenkins_mcu/workspace/avr8-gnu-toolchain-osx/avr8-gnu-toolchain-darwin_x86_64-hostlibs --with-mpc=/Volumes/build_disk/jenkins_mcu/workspace/avr8-gnu-toolchain-osx/avr8-gnu-toolchain-darwin_x86_64-hostlibs --with-pkgversion=AVR_8_bit_GNU_Toolchain_3.7.0_518 --with-bugurl=http://www.microchip.com
Thread model: single
gcc version 7.3.0 (AVR_8_bit_GNU_Toolchain_3.7.0_518) 
COLLECT_GCC_OPTIONS='-Os' '-g'  '-v' '-specs=device-specs/specs-attiny10' '-mmcu=avrtiny'

 /usr/local/avr8-Microchip-3.7.0/bin/../libexec/gcc/avr/7.3.0/cc1plus -quiet -v -imultilib avrtiny -iprefix /usr/local/avr8-Microchip-3.7.0/bin/../lib/gcc/avr/7.3.0/ -D__AVR_ATtiny10__ -D__AVR_DEVICE_NAME__=attiny10 foo.cpp -mn-flash=1 -mno-skip-bug -mabsdata -quiet -dumpbase foo.cpp -mmcu=avrtiny -auxbase foo -g -Os -version -mn-flash=1 -mno-skip-bug -mabsdata -fno-rtti -fno-enforce-eh-specs -fno-exceptions -o /var/folders/gv/zn3wcml52jq0vvjnd95j8g6h0000gp/T//ccgdAOQI.s

GNU C++14 (AVR_8_bit_GNU_Toolchain_3.7.0_518) version 7.3.0 (avr)
	compiled by GNU C version 5.5.0, GMP version 5.0.2, MPFR version 3.0.0, MPC version 0.9, isl version none
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "/usr/local/avr8-Microchip-3.7.0/bin/../lib/gcc/avr/7.3.0/../../../../avr/include/c++/7.3.0"
ignoring nonexistent directory "/usr/local/avr8-Microchip-3.7.0/bin/../lib/gcc/avr/7.3.0/../../../../avr/include/c++/7.3.0/avr/avrtiny"
ignoring nonexistent directory "/usr/local/avr8-Microchip-3.7.0/bin/../lib/gcc/avr/7.3.0/../../../../avr/include/c++/7.3.0/backward"
ignoring nonexistent directory "/usr/local/avr8-Microchip-3.7.0/bin/../lib/gcc/avr/7.3.0/../../../../avr/sys-include"
ignoring nonexistent directory "/usr/local/avr8-Microchip-3.7.0/bin/../lib/gcc/../../lib/gcc/avr/7.3.0/../../../../avr/include/c++/7.3.0"
ignoring nonexistent directory "/usr/local/avr8-Microchip-3.7.0/bin/../lib/gcc/../../lib/gcc/avr/7.3.0/../../../../avr/include/c++/7.3.0/avr/avrtiny"
ignoring nonexistent directory "/usr/local/avr8-Microchip-3.7.0/bin/../lib/gcc/../../lib/gcc/avr/7.3.0/../../../../avr/include/c++/7.3.0/backward"
ignoring duplicate directory "/usr/local/avr8-Microchip-3.7.0/bin/../lib/gcc/../../lib/gcc/avr/7.3.0/include"
ignoring duplicate directory "/usr/local/avr8-Microchip-3.7.0/bin/../lib/gcc/../../lib/gcc/avr/7.3.0/include-fixed"
ignoring nonexistent directory "/usr/local/avr8-Microchip-3.7.0/bin/../lib/gcc/../../lib/gcc/avr/7.3.0/../../../../avr/sys-include"
ignoring duplicate directory "/usr/local/avr8-Microchip-3.7.0/bin/../lib/gcc/../../lib/gcc/avr/7.3.0/../../../../avr/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/local/avr8-Microchip-3.7.0/bin/../lib/gcc/avr/7.3.0/include
 /usr/local/avr8-Microchip-3.7.0/bin/../lib/gcc/avr/7.3.0/include-fixed
 /usr/local/avr8-Microchip-3.7.0/bin/../lib/gcc/avr/7.3.0/../../../../avr/include
End of search list.
GNU C++14 (AVR_8_bit_GNU_Toolchain_3.7.0_518) version 7.3.0 (avr)
	compiled by GNU C version 5.5.0, GMP version 5.0.2, MPFR version 3.0.0, MPC version 0.9, isl version none
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: 1d343ef2b854a15058333158d7755550
COLLECT_GCC_OPTIONS='-Os' '-g'  '-v' '-specs=device-specs/specs-attiny10' '-mmcu=avrtiny'
 /usr/local/avr8-Microchip-3.7.0/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/as -mmcu=avrtiny -mno-skip-bug -o /var/folders/gv/zn3wcml52jq0vvjnd95j8g6h0000gp/T//ccqqOUkc.o /var/folders/gv/zn3wcml52jq0vvjnd95j8g6h0000gp/T//ccgdAOQI.s
[... cut ...] 3.6.1-arduino7/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/libc.a when searching for -lc
/Applications/Arduino-1.8.13.app/Contents/Java/portable/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: cannot find -lc
/Applications/Arduino-1.8.13.app/Contents/Java/portable/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: cannot find -lattiny10
collect2: error: ld returned 1 exit status

```