arduino/toolchain-avr #66: [328PB] Missing definitions in power.h

hitech95, 2019-11-18:
I thought it was a problem of MiniCore but after some throubleshooting its not.
[#117](https://github.com/MCUdude/MiniCore/issues/117)

So far I have found that thoose functions are missing:
```
  power_spi_disable();
  power_twi_disable();
  power_adc_disable();
```
But I think that others might also missing. The same code compiled for the 168PB works fine.

I'm using Arduino 1.8.0 that is shipped with:
```
arduino.avrdude=6.3.0-arduino17
arduino.arduinoOTA=1.3.0
arduino.avr-gcc=7.3.0-atmel3.6.1-arduino5
```

MCUdude, 2019-11-18:
Just for reference, power.h does not work with ATmega324PB either.

awatterott, 2020-05-03:
The following things have to be changed in order to get the power functions working for m324PB and m328PB controller:

power.h
add `__AVR_ATmega324PB__` and `__AVR_ATmega328PB__` defines

iom324pb.h and iom328pb.h
`PRSPI0` -> `PRSPI`
`PRTWI0 `-> `PRTWI`
`__AVR_HAVE_PRR0_PRSPI0` -> `__AVR_HAVE_PRR0_PRSPI`
`__AVR_HAVE_PRR0_PRTWI0` -> `__AVR_HAVE_PRR0_PRTWI`

SpenceKonde, 2020-05-03:
Huh? That seems like a bad approach - that IS correctly defined in the io header - there are two SPI and TWI peripherals, and as far as I can tell they both get a bit in a PRR register to shut them down... So we should instead leave iom32*pb.h untouched, and add the following to power.h so that it can understand the real register names - and will provide the power management functions for the second SPI and TWI peripherals too

#if defined(__AVR_HAVE_PRR0_PRSPI0)
#define power_spi_enable()              (PRR0 &= (uint8_t)~(1 << PRSPI0))
#define power_spi_disable()             (PRR0 |= (uint8_t)(1 << PRSPI0))
#endif

#if defined(__AVR_HAVE_PRR1_PRSPI1)
#define power_spi1_enable()              (PRR1 &= (uint8_t)~(1 << PRSPI1))
#define power_spi1_disable()             (PRR1 |= (uint8_t)(1 << PRSPI1))
#endif

#if defined(__AVR_HAVE_PRR2_PRSPI1)
#define power_spi1_enable()              (PRR2 &= (uint8_t)~(1 << PRSPI1))
#define power_spi1_disable()             (PRR2 |= (uint8_t)(1 << PRSPI1))
#endif

#if defined(__AVR_HAVE_PRR0_PRTWI0)
#define power_twi_enable()              (PRR0 &= (uint8_t)~(1 << PRTWI0))
#define power_twi_disable()             (PRR0 |= (uint8_t)(1 << PRTWI0))
#endif

#if defined(__AVR_HAVE_PRR1_PRTWI1)
#define power_twi1_enable()             (PRR1 &= (uint8_t)~(1 << PRTWI1))
#define power_twi1_disable()            (PRR1 |= (uint8_t)(1 << PRTWI1))
#endif

#if defined(__AVR_HAVE_PRR2_PRTWI1)
#define power_twi1_enable()             (PRR2 &= (uint8_t)~(1 << PRTWI1))
#define power_twi1_disable()            (PRR2 |= (uint8_t)(1 << PRTWI1))
#endif

And as you said, add those parts to the list...

awatterott, 2020-05-03:
Yes, that is also possible. I though to use the existing definitions from the other controllers, but you are right, because` __AVR_HAVE_PRR1_PRSPI1` and `__AVR_HAVE_PRR1_PRTWI1` is also missing in `power.h`.

chaveiro, 2020-05-03:
@SpenceKonde some typos that but it's that the path.

SpenceKonde, 2020-05-03:
Yeah, I was doing it in github off the top of my head *shrug*

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

chaveiro, 2020-05-03:
Unfortunately i'm not sure where is the repository for  ** hardware\tools\avr\avr\include\avr\power.h ** file.
Is it part of AVT toolchain from Atmel (now Microship) ?
https://www.microchip.com/mplab/avr-support/avr-and-arm-toolchains-c-compilers

hitech95, 2020-05-31:
I've seen that some changes has been merged.
Are those available to general public?

If not how can I test the new patches?

SpenceKonde, 2020-09-25:
These fixes are in arduino7 version of toolchain, which is included by default on 1.8.13. 

This issue should be closed.

hitech95, 2020-09-25:
Just tested, its working.

commit 9f8c0dc2 (2020-05-26): Materials for #66, #67, and another issue

See info.txt in alsoneeded directory for more information on this.

These files need to replace the standard versions of them in order to
fix #66, #67, and an issue with eeprom_is_ready() in eeprom.h that
manifests on the newer parts. Unfortunately I'm not good enough at shell
scripting to make the build scripts slip these fixes in
closed 2020-09-25 by hitech95