arduino/toolchain-avr #67: Not able to compile for ATmega3209/3208/1609/1608/809/808 using the latest toolchain

MCUdude, 2020-04-27:
I stumbled across this weird issue today. I'm not able to build for any of the mentioned targets using the latest 7.3.0-atmel3.6.1-arduino6 toolchain. It does work for ATmega4809/4808 though. This is the error I'm getting:

```
/Users/hans/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino6/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: address 0x803066 of /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_493085/230V_trigger.ino.elf section `.bss' is not within region `data'
/Users/hans/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino6/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: address 0x803066 of /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_493085/230V_trigger.ino.elf section `.bss' is not within region `data'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ATmega3209.
```

It works flawlessly with 7.3.0-atmel3.6.1-arduino5. Any idea what may cause this?
If you want to try it yourself, you can install [MegaCoreX](https://github.com/MCUdude/MegaCoreX) 1.0.2 from the boards manager. Note that 1.0.2 has a bug that prevents the 32-pin pinout from building (will fix it very soon). If you want to test this on ATmega3208/6108/808 you will have to select the 28-pin pinout.

SpenceKonde, 2020-04-27:
That error sounds super familiar.... IIRC it was also something silly - like, it didn't stick in my head enough for me to remember the details

MCUdude, 2020-04-28:
@SpenceKonde something silly with the toolchain, or something stupidly simple you had to do to fix it?

SpenceKonde, 2020-04-28:
Yeah, one of those. Actually, I think what I wound up doing (because I
couldn't get crosscompile working anyway so actually building all the stuff
wasnt going to work), that worked for now, was a bastardized toolchain
pack, with arduino5 binaries, but latest libraries from atpacks. Didnt fet
far enough along to convert that into packages though

Messaging you on my slack (fate intervened and I'm not where I want to be
on several fronts currently) things are moving, just slower

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

MCUdude, 2020-04-29:
@facchinm do you know what's causing this? Are there other changes to *arduino6 other than an updated pack?

SpenceKonde, 2020-05-01:
In recent ATpacks, DATAMEM_SIZE has changed - looking at the latest atpacks, almost everything is else is a simple fix or correction, plus some changed to how reserved registers are shown... but DATAMEM_SIZE has grown, and for many chips is now 2^16. I can't say I know exactly what this macro means - but I'm pretty sure this change is what's making everything barf here....

MCUdude, 2020-05-01:
This seems to be a known issue with the latest pack. It's caused by the linker script

[https://www.avrfreaks.net/forum/failing-linker-new-device-pack-attiny-0-series](https://www.avrfreaks.net/forum/failing-linker-new-device-pack-attiny-0-series)

MCUdude, 2020-05-01:
this link explains what went wrong:

[https://www.avrfreaks.net/comment/2803036#comment-2803036](https://www.avrfreaks.net/comment/2803036#comment-2803036)

MCUdude, 2020-09-07:
Whoops, forgot to close this!

closed 2020-09-07 by MCUdude