arduino/toolchain-avr #69: GCC version in README out of date

matthijskooijman, 2020-05-19:
The README talks about gcc 5.4.0, but I think the Atmel 3.6.1 pack contains gcc 7.3.0. Haven't checked the other version numbers.

matthijskooijman, 2020-05-19:
Hm, I'm a bit confused, since Debian has a `1:5.4.0+Atmel3.6.1-2` version of gcc-avr: https://packages.debian.org/buster/gcc-avr but the Arduino AVR JSON file has

```
            {
              "size": "39114120",
              "checksum": "SHA-256:3effed8ffa1978b6e4a46f1aa2acc629e440b4d77244f71f9b79a916025206fb",
              "host": "x86_64-linux-gnu",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-arduino5-x86_64-pc-linux-gnu.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-x86_64-pc-linux-gnu.tar.bz2"
            }
```

which suggests that 3.6.1 contains 7.3.0...

facchinm, 2020-05-22:
It's a bit tricky :slightly_smiling_face: 
Atmel toolchain 3.6.1 contained gcc 5.4.0 + a set of patches on top.
These patches were extracted and backported here https://github.com/arduino/toolchain-avr/tree/staging/avr-gcc-patches 
I didn't know how to call the toolchain otherwise, which can be a bit confusing indeed. Any hint is appreciated :wink:

closed 2020-05-22 by facchinm

matthijskooijman, 2020-05-22:
Right, but I meant mostly this section of the README:

https://github.com/arduino/toolchain-avr/blob/staging/Readme.md#configuring

I guess that it might technically just say that Atmel 3.6.1 contains gcc 5.4.0, but to me it looks like the current toolchain-avr-produced package contains gcc 5.4.0, which is not true.

matthijskooijman, 2020-05-22:
Maybe an explicit note in the README that says that the atmel package was based on 5.4.0, but was updated to apply to 7.something would be helpful here? Maybe also for other versions that were bumped, if any?

facchinm, 2020-05-22:
Correct, let me amend the README immediately