arduino/toolchain-avr #76: distribute.atmel.no : Broken link in documentation.

blogarithme, 2020-12-22:
Hi,

The link http://distribute.atmel.no/tools/opensource/Atmel-AVR-GNU-Toolchain/ used in the file `Readme.md` on the words ‘_ship a newer toolchain_’ and on ‘_latest version is._’  is invalid (Timeout).

The replacement link could be https://www.microchip.com/en-us/development-tools-tools-and-software/gcc-compilers-avr-and-arm ?

blogarithme, 2020-12-24:
The text on the page https://www.microchip.com/en-us/development-tools-tools-and-software/gcc-compilers-avr-and-arm looks good but all the links related to AVR (excepted the last one) are broken. They point to w3schools !!!

hlovdal, 2022-02-06:
The link is now <https://www.microchip.com/en-us/tools-resources/develop/microchip-studio/gcc-compilers>. And yes, getting this fixed would be good.