arduino/toolchain-avr #77: Windows build: gdb.build.bash needs undocumented dependencies.

blogarithme, 2020-12-23:
Hi,

I’m trying to build the AVR Toolchain from, and for, Windows, but I get some issues during the execution of `gdb.build.bash`
Here what I did:
1) Setup:
- I installed Cygwin (32bits) and added the packages as explained on the front page of this repo, excepted for `texinfo` because the Cygwin setup only provides version 6. Version 4 is no more available. 
- I installed MinGW.
- I cloned this repo.

2) Execution:
From a Cygwin console I executed, one by one, the bash scripts in the order given in the documentation:

```
./tools.bash             # ok
./binutils.build.bash    # ok
./gcc.build.bash         # ok (very long)
./avr-libc.build.bash    # ok 
./gdb.build.bash         # aborted due to warnings and error
```
Here the last lines displayed

> ```
> …
> checking for shared library run path origin... done
> checking for libexpat... no
> configure: WARNING: expat is missing or unusable; some features may be unavailable.
> checking whether to use python... auto
> checking for python... missing
> checking for python2.7... no
> checking for python2.6... no
> checking for python2.5... no
> checking for python2.4... no
> configure: WARNING: python is missing or unusable; some features may be unavailable.
> checking whether to use guile... guile-2.0
> checking for pkg-config... /usr/bin/pkg-config
> checking for usable guile from /usr/bin/pkg-config... configure: error: unable to find usable guile version from "guile-2.0"
> make[1]: *** [Makefile:9614: configure-gdb] Error 1
> make[1]: Leaving directory '/c/toolchain-avr-master/gdb-build'
> make: *** [Makefile:836: all] Error 2
> ```

To fix these warnings and errors (libexpat, python, guile) I added these 7 packages to my Cygwin setup (in addition to those listed in the documentation):

  libexpat-devel 
  python2
  python27-devel
  libintl-devel 
  guile2.0 
  libguile2.0-devel 
  libgmp-devel 

Here the summary of all the packages I explicitly added to the default Cygwin setup.

[image]

I hope this can be added to the documentation in the "Windows requirements" section.

Regards.

facchinm, 2020-12-23:
Hi @blogarithme ,
would you mind filing a PR with the relevant updates/changes? It's been a while since we last built the windows toolchain with cygwin (we are currently cross-compiling from linux using mingw) but updated instruction are always useful.

blogarithme, 2020-12-23:
Hi @facchinm 
I first make an overview of all the problems I encounter with Cygwin and, if the _experts_ don't detect any nonsense in my solutions, I will try my first PR.

DamianReloaded, 2021-06-20:
This applies also to Debian 10.8 . They should probably be added in the "Debian requirements" section of readme.md

libexpat1-dev 
python2 
python2.7-dev 
libint-dev 
guile-2.0-dev 
guile-2.0-libs 
libgmp-dev

After adding these dependencies I was able to build gdb using gdb.build.bash (otherwise it gives Make Error 2). Maybe some of those aren't actually required. It works nonetheless.