arduino/toolchain-avr #88: `error: value '64' out of range for constraint 'I'` in `avr/include/avr/wdt.h`

KOLANICH, 2022-06-07:
https://github.com/arduino/toolchain-avr/issues

When tryint to setup a CMake toolchain file to compile to avr target using `clang-15` I got the following errors:

```
In file included from ./1.8.19/hardware/arduino/avr/cores/arduino/CDC.cpp:20:
./1.8.19/hardware/tools/avr/lib/gcc/avr/7.3.0/../../../../avr/include/avr/wdt.h:464:12: error: value '64' out of range for constraint 'I'
                                : "I" (_SFR_IO_ADDR(_WD_CONTROL_REG)),
                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

(in `wdt_disable`)

```
In file included from ./1.8.19/hardware/arduino/avr/cores/arduino/CDC.cpp:20:
./1.8.19/hardware/tools/avr/lib/gcc/avr/7.3.0/../../../../avr/include/avr/wdt.h:507:23: error: value '64' out of range for constraint 'I'
                                : [WDTREG]  "I"  (_SFR_IO_ADDR(_WD_CONTROL_REG)),
                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

(in `wdt_disable`)

I guess 
1. `& 0b111111` is needed for runtime dispatch to calm down the compiler.
2. I also wonder if we somehow can utilize `if constexpr` for the case of C++ to do dispatch in compile time.

SpenceKonde, 2022-06-07:
Yer not trying to compiler that library on a modern/post2016 AVR are you? That won't work, they haven't updated the library.

KOLANICH, 2022-06-07:
>Yer not trying to compiler that library on a modern/post2016 AVR are you?

The mcu I compile for is atmega2560.

>That won't work, they haven't updated the library.

What do you mean? Should not clang link well to any standard library gcc links well? 

If I understand right, the issue mentioned here can happen if clang has stricter checks than gcc has (and it is often tye case).

SpenceKonde, 2022-06-08:
No library that uses _SFR_TO_IO_ADDR on a special function register that doesn't reside in the I/O spacve will ever work. 

And yeah apparently that's the case on the 2560. It's 0x60. But 0x3F is the highest number allowed for the I constraint, which specifies the address of a locatrion within the I/O space. On the 2560, the WDT control registrer is not located within the I/O space, and the assembly that is being used to access it cannot work.

KOLANICH, 2022-06-08:
Thanks for the info.

1. I guess I have mistaken, `arduino/toolchain-avr` is the wrong repo to fill an issue against. I probably should have created the issue in https://github.com/avrdudes/avr-libc

2. The func has 2 branches: https://github.com/avrdudes/avr-libc/blob/960a359d854001399b8fb445a9ad0ce3ea80f4d0/include/avr/wdt.h#L450-L532 , one uses `out` and another one uses `sts`, I guess on this board the one using `sts` is expected to be taken, but the content of `if` is not a constexpr, and if I rewrite it to `if constexpr` it errors because [`#define _SFR_MEM_ADDR(sfr) ((uint16_t) &(sfr))`](https://github.com/avrdudes/avr-libc/blob/502f5091d2b49191a87eb4a3a926525a2a34926f/include/avr/sfr_defs.h#L182-L184) is not a constexpr.

KOLANICH, 2022-06-08:
The issue has already been filed by someoje else there: https://github.com/avrdudes/avr-libc/issues/678

closed 2022-06-08 by KOLANICH