arduino/toolchain-avr #91: toolchain-avr for Apple silicon (aarch64, arm64)?

imrehorvath, 2022-08-21:
Hello!
I know we should be able to use the `avr-gcc-7.3.0-atmel3.6.1-arduino7-x86_64-apple-darwin14` with Rosetta 2 on Apple silicon Macs for now, but I'm curious if it's considered to provide a native build for Apple silicon Macs for the near or far future.

Ekkehard, 2026-08-29:
While the arm64 support is still brewing, here is a script that installs arm64 binaries via brew and lets arduino-cli and friends believe they are its own (replace x85_64 binaries with links to the arm64 ones: [https://github.com/Ekkehard/arduino-cli-go-native](url).  Hope it helps somebody.

PaulStoffregen, 2026-11-11:
Another AVR toolchain build for Apple Silicon has been made by Matthew Yang.

https://github.com/matthewyang204/arduino-1x-toolchain-avr